microsoft/vscode-python-devicesimulator #155: Can't see serial output

roblourens, 2019-11-09:
I picked up the adafruit device at Ignite and I assume this is where issues should go, let me know if there is somewhere else you want them.

I am trying to open the serial console. When I run "Select serial port", nothing happens and there is no UI feedback. But I do see this in the developer tools:

```
no library available after trying files [ long list of files ]

stack trace: TypeError: Cannot read property 'close' of null	at Object.<anonymous> (/Users/roblou/.vscode-insiders/extensions/ms-python.devicesimulatorexpress-2019.0.35146/vendor/node-usb-native/lib/bindings.js:27:19)	at Module.o._compile (/Applications/Visual Studio Code - Insiders.app/Contents/Resources/app/out/vs/loader.js:15:749)	at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)	at Module.load (internal/modules/cjs/loader.js:645:32)	at Function.Module._load (internal/modules/cjs/loader.js:560:12)
```

It did prompt me to install dependencies earlier and it seems that didn't happen. Ideally it would show me a nicer error and this unhandled exception wouldn't happen.

iennae, 2019-11-10:
This is exactly where issues should go. Thanks for this feedback and the follow-up issue.

andreamah, 2020-02-11:
Fixed in #194 . Thanks for letting us know! 🚀

closed 2020-02-11 by andreamah