microsoft/vscode-python-devicesimulator #156: Include repo in package.json for issues

roblourens, 2019-11-09:
You should link to this repo in the package.json to make it easier for people to file issues or submit PRs:

```
  "repository": {
    "type": "git",
    "url": "https://github.com/microsoft/vscode-python-devicesimulator.git"
  },
  "bugs": {
    "url": "https://github.com/microsoft/vscode-python-devicesimulator/issues"
  },
```

DonJayamanne, 2019-11-10:
Landed a PR to resolve this https://github.com/microsoft/vscode-python-devicesimulator/pull/159

roblourens, 2019-11-11:
Thanks! I would have sent a PR but I was on a very bad connection at the time...

vandyliu, 2020-01-16:
Issue fixed in [PR159](https://github.com/microsoft/vscode-python-devicesimulator/pull/159). Marking it as closed, let us know if you encounter any other issue.

closed 2020-01-16 by vandyliu