microsoft/vscode-python-devicesimulator #157: Show error when pip fails

roblourens, 2019-11-09:
I realized that `pip` is not installed and when the extension tries to install its dependencies, it silently fails. I just have this in the developer tools log:

```
Error: Command failed: python3.7 -m pip install -r /Users/roblou/.vscode-insiders/extensions/ms-python.devicesimulatorexpress-2019.0.35146/out/requirements.txt -t /Users/roblou/.vscode-insiders/extensions/ms-python.devicesimulatorexpress-2019.0.35146/out/python_libs/usr/local/bin/python3.7: No module named pip
```

Please show an error popup when this fails, I think this is why #155 happened

roblourens, 2019-11-09:
Also with pip installed correctly, I get this error:

```
rror: Command failed: python3.7 -m pip install -r /Users/roblou/.vscode-insiders/extensions/ms-python.devicesimulatorexpress-2019.0.35146/out/requirements.txt -t /Users/roblou/.vscode-insiders/extensions/ms-python.devicesimulatorexpress-2019.0.35146/out/python_libsERROR: Could not find a version that satisfies the requirement pywin32==224 (from -r /Users/roblou/.vscode-insiders/extensions/ms-python.devicesimulatorexpress-2019.0.35146/out/requirements.txt (line 6)) (from versions: none)ERROR: No matching distribution found for pywin32==224 (from -r /Users/roblou/.vscode-insiders/extensions/ms-python.devicesimulatorexpress-2019.0.35146/out/requirements.txt (line 6))
```

Maybe because I am on a Mac. I deleted that line and it seems to have installed them fine after that.

vandyliu, 2020-02-22:
Fixed in https://github.com/microsoft/vscode-python-devicesimulator/pull/210

closed 2020-02-22 by vandyliu