microsoft/vscode-python-devicesimulator #158: Extension logs noisily

roblourens, 2019-11-09:
- Install and run the extension
- Open developer tools (Help > Toggle Developer Tools)
- See pages of log messages:

[image]

- Please only log if necessary. You can also respect vscode's log level or add a setting if you want to enable logging. Or you could log to an update channel (but still be careful about producing very large files)

LukeSlev, 2019-11-10:
Hey @roblourens, looking at those logs it looks like those are actually coming from the Python extension which our extension depends on. As a result we don't have too much control over the level of logging that they do. However, you could always post an issue on the Python repo! https://github.com/microsoft/vscode-python

DonJayamanne, 2019-11-10:
Agreed, its our Python extension that's logging this info. Can be closed in favor of https://github.com/microsoft/vscode-python/issues/2156

closed 2019-11-10 by DonJayamanne

roblourens, 2019-11-10:
Sorry, I didn't look closely