microsoft/vscode-python-devicesimulator #163: Investigate issue(s) reported on the marketplace review with regards to virtual environments and pythonPath setting

iennae, 2019-11-15:
From review on the marketplace

`Extension currently does not work with python virtual environments and ignores the "python.pythonPath"  value in settings.  Therefore all python modules must be installed globally, which is something I'm not willing to do for a beta extension.`

gramster, 2020-03-05:
Related, I got this today. I wasn't using the extension but I had just restarted VS Code. I'm guessing this is because I am on MacOS and it looks at my system Python rather than my virtual environments (where I was on 3.8). This is something we need to fix but it can be postponed a little bit while the Python extension team do some reworking of the environment API which should make the fix easier. 

[image]

iennae, 2020-03-05:
@gramster  I was able to select a python virtual environment and not install modules globally. I am on MacOS as well. It sounds like you replicated something similar to the original reporter. Do you have more information about what could be causing the problem?

andreamah, 2020-03-05:
Hi @gramster ! 

It's strange how it happened when you weren't even using the extension, but I can maybe see why you got the error although you had a virtual environment selected. Currently, the extension uses the "python.pythonPath" from the Python extension in order to determine the version to use. However, when you first open VS Code with nothing previously set in a workspace or global settings.json (usually when you don't have a .vscode workspace folder and the global settings.json doesn't have python.pythonPath pre-defined), it just uses "python", which is the system Python. At the same time, the interpreter listed in the bottom left corner upon startup is a different from the system version (likely the newest on the machine). 

[image]

Therefore, when we open the extension, it mistakenly uses the system Python from the Python extension and shows that the version is too low. We'll look into a way to see if we can work around this in a future release.

Thanks for the feedback! :)

andreamah, 2020-04-18:
Addressed in #338 . If the Python path is not specifically set, it will search for the version of Python that is under "python3" (if available). This should fix the issue 😊.

I will wait one business day and close this if there are no further comments.

andreamah, 2020-04-21:
Closing because of no further feedback.

closed 2020-04-21 by andreamah