microsoft/vscode-python-devicesimulator #166: Show errors when running linting

rvanlaar, 2019-12-10:
My linter ran into problems. It crashed during the run. 
I noticed due to me not seeing any linting errors when there should be some.

VSCode didn't give any indication that something was off.

There was actuall no way in VScode to see these errors.
Since the Python Log only seems to show stdout.

I would expect it work just like the discovery of tests in VSCode.
These show a warning error when the discovery fails.

andreamah, 2020-03-13:
Hi @rvanlaar,

Thanks for reaching out, and apologies for the late reply. Could you outline some steps to replicate this issue?

Thanks!

rvanlaar, 2020-03-13:
Thank you for getting back to me.

Am I in the right project though? I see that this one is vscode-python-devicesimulator.
My issue is with the vscode-python plugin.

To reproduce. Put a 'raise' in the pylint program, so that it crashes.
And run it via vscode. 
VScode won't show any errors regarding the crashed program. 

I had two problems, one was that pipenv installed incompatible packages, which made pylint crash,
another one which was this one: https://github.com/microsoft/vscode-python/issues/9661

vandyliu, 2020-03-13:
It looks like this issue is in the wrong place. Pylint issues with are likely due to the vscode-python extension and should be put here: https://github.com/microsoft/vscode-python/. They are a great team and I have a good feeling they will be able to help you out more than we can. If you have any issues with our extension, feel free to open up a new issue or comment here. 

Best of luck!

closed 2020-03-13 by vandyliu