microsoft/vscode-python-devicesimulator #249: Can the CircuitPython support be broken out?

jimbobbennett, 2020-03-06:
There are a number of devices that support CircuitPython from Adafruit, such as the [PyBadge](https://www.adafruit.com/product/4200) or [PyPortal](https://www.adafruit.com/product/4116).

It would be great to have limited VS Code support for programming these devices including:

* Intellisense and code completion for CircuitPython code
* Integrated serial monitor
* Code deployment

Rather than focus on adding all these devices to this simulator, could this extension be split in two? One part providing the capabilities for using CircuitPython with any Adafruit device, and the other part adding the simulator?

smmatte, 2020-03-10:
Hi @jimbobbennett this is a great idea. For the short term we don't have plans to do this work ourselves.

Josverl, 2020-05-15:
@jimbobbennett 
I suggest that you taske a look at 
- [pymakr](https://marketplace.visualstudio.com/items?itemName=pycom.Pymakr) 
  this add-in allows you work with hardware genric and pycom boards 
- [micropy-cli](https://pypi.org/project/micropy-cli/) 
which adds micropython intellisense (for any firmware) to you projects using 
- [Micropython-stubber](https://github.com/josverl/micropython-stubber)
  a project that i created to help address my own need for the issues yo mentioned.

gramster, 2020-08-26:
We do have a long term vision for this project to be more easily extensible to new devices. However, as an intern project (currently without interns) its unlikely to happen soon.