microsoft/vscode-python-devicesimulator #268: Sensor Implementation Requests (Vote in comments for which features you would like for us to support)

vandyliu, 2020-03-19:
👍 (+1)  the sensors/features you would like for us to implement in the comments.

vandyliu, 2020-03-19:
IR Sensor for the Circuit Playground Express

vandyliu, 2020-03-19:
Tap and Double Tap feature in the Motion Sensor for the Circuit Playground Express

vandyliu, 2020-03-19:
Sound Sensor for the Circuit Playground Express

vandyliu, 2020-03-26:
Compass Sensor for micro:bit

vandyliu, 2020-03-26:
Sound Sensor for micro:bit

askpatrickw, 2020-05-06:
GPIO Pins for the Circuit Playground Express

askpatrickw, 2020-05-07:
play_tone, start_tone, stop_tone for the CP Express

s3111560, 2020-06-12:
compass sensor for micro:bit

pinarodrigo, 2020-07-12:
Bluetooth & Radio Sensor for microbit

UnicycleDumpTruck, 2020-08-14:
Sound output (buzzer) for the CLUE! please :)

silvasteve110, 2020-08-24:
(+1) speaker for circuit playground express

jfield12050, 2020-09-11:
Sound Sensor for the CLUE, please!

m-farkas, 2020-12-28:
GPIO (Pins) for microbit