microsoft/vscode-python-devicesimulator #277: VSCODE CPX Simulator not working

jac141, 2020-03-24:
Hello

Everytime i hit "cmd + shift + P" to open a "Device Simulator Express: [Circuit Playground Express] New File" it tells me that the command does not exist. In fact though I have made sure to install the extension and all the dependencies none of the commands associated with the device simulator work. I have tried reinstalling multiple times. I am working on ubuntu 18.04. I am not sure what to do since the commands are not working.

Need help with this given the coronavirus keeping everyone out of the lab.

xnkevinnguyen, 2020-03-24:
Hi Jac,
This issue happens when the VS code python extension is not installed or not activated. Since we rely on this extension it will be needed to use our simulator. Let us know if this works for you! 

Thanks for using our extension!

askpatrickw, 2020-03-25:
@xnkevinnguyen can you provide a link to that extension? The reason I ask is I am also seeing this issue. I DO have the [Python extension published by Microsoft](https://marketplace.visualstudio.com/items?itemName=ms-python.python) installed.

```
Version: 1.43.2
Commit: 0ba0ca52957102ca3527cf479571617f0de6ed50
Date: 2020-03-24T07:34:57.037Z
Electron: 7.1.11
Chrome: 78.0.3904.130
Node.js: 12.8.1
V8: 7.8.279.23-electron.0
OS: Darwin x64 19.3.0
```

jac141, 2020-03-25:
Hello,

How do we activate the VS code python extension? I am pretty sure that I
have it installed. If I do not have it installed how do I do so. I
followed the instructions to install it but it does not work.

Cheers,
Jordan Campbell
UCSD PhD Candidate
4047078440
jac141@eng.ucsd.edu

askpatrickw, 2020-03-26:
> @xnkevinnguyen can you provide a link to that extension? The reason I ask is I am also seeing this issue. I DO have the [Python extension published by [...]

SO.... no I wasn't ;-)
I was hitting ⌘P not ⇧⌘P so I was not at the right prompt.
If DO use ⌘P and then > I can bring up the Device Simulator commands.
Same things if I use ⇧⌘P.

jac141, 2020-03-26:
Hello,

I am hitting the right buttons and it is not working.

xnkevinnguyen, 2020-03-26:
Hi Jordan, 
This is the link to the python extension if you don't have it already installed. https://marketplace.visualstudio.com/items?itemName=ms-python.python

Can you let me know which Python version you're using and if you have pip.

jac141, 2020-03-26:
Hello,

I have 3.7 and I installed pip on my computer

jac141, 2020-03-27:
Hello

I have all the right stuff installed.

Cheers,
Jordan Campbell
UCSD PhD Candidate
4047078440
jac141@eng.ucsd.edu

vandyliu, 2020-03-27:
Are you using the extension from the marketplace or trying to compile from the source code?

jac141, 2020-03-27:
I don't know I downloaded the vs code. Installed it and tried to open it. I
am just following the instructions I Saw

xnkevinnguyen, 2020-03-27:
Ok, we have identified the issue on Ubuntu. We are currently working on a fix. We'll give you an update when it is fixed. This will most likely occur next week. 

Again, Thank you for using our extension and for your patience :)

xnkevinnguyen, 2020-03-31:
Fix has been deployed to production version. Let us know if it has been fixed for your issue.

jac141, 2020-03-31:
Hello Kevin,

It seems to be re-installing. Does it matter that it says preview next to
it or not.

Cheers,
Jordan Campbell
UCSD PhD Candidate
4047078440
jac141@eng.ucsd.edu

jac141, 2020-03-31:
Hello Kevin,

Well first I got a new error saying that I didn't have the right python
interpreter. And it says install 3.7+ and I have 3.7.7 as it shows in the
picture.

Cheers,
Jordan Campbell
UCSD PhD Candidate
4047078440
jac141@eng.ucsd.edu

xnkevinnguyen, 2020-04-02:
Closed as this has been communicated to have been fixed for Ubuntu

closed 2020-04-02 by xnkevinnguyen