microsoft/vscode-python-devicesimulator #332: Version 2020.0.36197 dependencies won't install error

profbrady, 2020-04-16:
I updated to the latest version today and cannot run the simulator. When dependencies are installed there is a failure. Falling back to the previous version works.

andreamah, 2020-04-16:
Hi @profbrady ,

I'm sorry to hear that! I just have a few questions:
- At which step does the extension fail?
- Which OS are you running? 
- Which version of VS Code do you have?
- Which version of Python are you running?

profbrady, 2020-04-16:
After installing the extension and reloading, I opened the simulator. I said yes to the message that asked if I wanted to install dependencies. Then I received this message 

"Dependency download could not be completed. Functionality may be limited. Please review the installation docs.”

So, I tried opening the simulator again and selected the CPX board. This time it opened, but when I try to run it I get the following long message in the output window.

[INFO] Deploying code to the simulator...
[INFO] File selected : z:\Downloads\cpy_3_04.py 

[ERROR] Traceback (most recent call last):
  File "c:\Users\bradyb\.vscode\extensions\ms-python.devicesimulatorexpress-2020.0.36197\out\check_python_dependencies.py", line 30, in check_for_dependencies
    pkg_resources.require(cleaned_dependencies)
  File "C:\Users\bradyb\AppData\Local\Programs\Python\Python37\lib\site-packages\pkg_resources\__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "C:\Users\bradyb\AppData\Local\Programs\Python\Python37\lib\site-packages\pkg_resources\__init__.py", line 786, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'adafruit-circuitpython-neopixel==5.0.0' distribution was not found and is required by the application

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\bradyb\.vscode\extensions\ms-python.devicesimulatorexpress-2020.0.36197\out\process_user_code.py", line 16, in <module>
    check_python_dependencies.check_for_dependencies()
  File "c:\Users\bradyb\.vscode\extensions\ms-python.devicesimulatorexpress-2020.0.36197\out\check_python_dependencies.py", line 32, in check_for_dependencies
    raise Exception(CONSTANTS.DEPEND_ERR)
Exception: The required dependencies aren't downloaded. Please use CTRL+SHIFT+P to open the command palette and select "Device Simulator Express: Install Extension Dependencies".

Running Windows 10 64-bit via Parallels on a MacBook Pro. Python 3.7.3. VS Code version 1.44.1

____
Brian

profbrady, 2020-04-16:
> Hi!
>
> Thanks for the detailed response. When you get the "Dependency download could not be completed. Functionality may be limited. Please review the installation docs.” message, is there also another pop-up asking to install the dependencies directly on your selected interpreter?

I will check again, but I do recall getting a message and choosing yes. I ran through the process with one of my students via Zoom just a few minutes ago. They were starting with a fresh Python and VS Code install and installed the extension. The latest one gav3 the same errors. Falling back to the previous version got them going. I have 41 other students who will be struggling with this issue this week and next. 

Brian

xnkevinnguyen, 2020-04-16:
Hi @profbrady ,

We are currently having difficulties replicating that exact issues when upgrading from the previous version to the current version on windows with the same python version you are using.

 Since it is our priority to get this working for you, having a quick call with you would be the fastest way to identify the issue and fix it. Let me know if you are available for a call today or tomorrow at this email: **t-xunguy@microsoft.com** . I will arrange a call with the devs of the extension.

vandyliu, 2020-04-16:
We have found the issue and are working on a fix. If you desperately need a fix, you can use a previous version of the extension or install git (which should fix the issue).

xnkevinnguyen, 2020-04-16:
@profbrady Should be fixed with the latest version on the market place with this hotfix 
https://github.com/microsoft/vscode-python-devicesimulator/pull/333. 

Let me know if the issue has been fixed on your side. I will close this issue in 24h otherwise.

profbrady, 2020-04-17:
It seems to be working. I have the latest version running successfully on Windows and Mac OS. Thanks for the quick turn around.

____
Brian

xnkevinnguyen, 2020-04-17:
Thank you for the update!

closed 2020-04-17 by xnkevinnguyen