microsoft/vscode-python-devicesimulator #342: Match virtual sensors ranges and value types to actual device

profbrady, 2020-04-19:
The CPX and CPB light sensors have a range of 0-320 but the simulator's range is 0-255. The CPX and CPB accelerometer values are floats so that you can actually read +/-9.8 for an axis when  stationary and the axis is vertical. Off the top of my head I don't recall what the temperature sensor range is on the physical device, but the value from it is a float whereas the simulator uses an integer. 

Something I would like to see regarding the accelerometer is the ability to change the rotation about an axis and have the associated acceleration values change in proportion to the amount of rotation. Rotation values of 0, 0, 0 would define an orientation of the board as if it were lying flat on a table with the buttons facing up. Or as seen on the screen with the +z axis facing the user. In this initial position the x and y drag bars would be set to zero but the z would be -9.8. This ability could be optional and turned on/off with a checkbox. 

The slide switch also seems to be backwards from this Adafruit guide.
https://learn.adafruit.com/circuitpython-made-easy-on-circuit-playground-express/slide-switch

Brian

xnkevinnguyen, 2020-04-19:
Thank you for the feedback Brian! I have created separate issues to keep track of individual bugs/suggestions. We will keep you updated on fixes and new features regarding each of those issues.

xnkevinnguyen, 2020-04-20:
This PR is addressing the sensor range value for light and we have added a one-decimal precision for values that are float on their input. #346 Will update when this PR will be on production.

closed 2020-04-20 by xnkevinnguyen

profbrady, 2020-04-20:
Great. Thanks a lot. 

Brian

xnkevinnguyen, 2020-04-23:
@profbrady The latest release has the sensor input changes for precision with 1 decimal. The CPX switch has also been inverted.

profbrady, 2020-04-23:
I saw that earlier today. Thank you very much. 

Brian