microsoft/vscode-python-devicesimulator #366: Getting started tab navigation

kattni, 2020-05-06:
This may be a limitation of VS Code.

When I open the Getting started tab, it provides a drop down menu to choose what I would like to "Learn more about:". I choose the board I am interested in, copy a code snippet, and click on my `code.py` file to paste the code snippet. When I return to the Getting Started tab it seems to reload, and return to the initial state with the drop down menu available. To get back to the code snippets, I must click on the menu and choose the board again each time.

If possible, consider allowing the Getting Started tab to remain in the chosen section so you can copy and paste multiple snippets of code without needing to choose the board from the drop down menu every time.

xnkevinnguyen, 2020-05-06:
This is a great suggestion, persisting the state is be possible. I'll assign myself to it :)

xnkevinnguyen, 2020-05-07:
should be persisting with #371

closed 2020-05-07 by xnkevinnguyen