microsoft/vscode-python-devicesimulator #368: Running simulator fails with ValueEerror: unknown locale: UTF-8

kattni, 2020-05-06:
To everyone who has worked on Device Simulator Express: Amazing job! I'm really glad to see you're continuing to maintain it.

I am having some issues getting it to run. I attempted to run the Device Simulator Express for Circuit Playground Express. The simulator will not run and it results in the following:
```
Welcome to the Device Simulator Express output tab!

[INFO] Deploying code to the simulator...
[INFO] File selected : /Users/kattni/Desktop/code.py 

[ERROR] Traceback (most recent call last):
  File "/Users/kattni/.vscode/extensions/ms-python.devicesimulatorexpress-2020.0.36321/out/process_user_code.py", line 45, in <module>
    from common.telemetry import telemetry_py
  File "/Users/kattni/.vscode/extensions/ms-python.devicesimulatorexpress-2020.0.36321/out/common/telemetry.py", line 35, in <module>
    telemetry_py = Telemetry()
  File "/Users/kattni/.vscode/extensions/ms-python.devicesimulatorexpress-2020.0.36321/out/common/telemetry.py", line 12, in __init__
    self.telemetry_client = TelemetryClient("__AIKEY__")
  File "/Users/kattni/.vscode/extensions/ms-python.devicesimulatorexpress-2020.0.36321/venv/lib/python3.7/site-packages/applicationinsights/TelemetryClient.py", line 30, in __init__
    self._context = channel.TelemetryContext()
  File "/Users/kattni/.vscode/extensions/ms-python.devicesimulatorexpress-2020.0.36321/venv/lib/python3.7/site-packages/applicationinsights/channel/TelemetryContext.py", line 38, in __init__
    self.device = contracts.Device()
  File "/Users/kattni/.vscode/extensions/ms-python.devicesimulatorexpress-2020.0.36321/venv/lib/python3.7/site-packages/applicationinsights/channel/contracts/Device.py", line 22, in __init__
    self._initialize()
  File "/Users/kattni/.vscode/extensions/ms-python.devicesimulatorexpress-2020.0.36321/venv/lib/python3.7/site-packages/applicationinsights/channel/TelemetryContext.py", line 14, in device_initialize
    self.locale = locale.getdefaultlocale()[0]
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/locale.py", line 568, in getdefaultlocale
    return _parse_localename(localename)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/locale.py", line 495, in _parse_localename
    raise ValueError('unknown locale: %s' % localename)
ValueError: unknown locale: UTF-8
```

I tried updating my Python environment globally to 3.7.6 on my computer (this version choice is arbitrarily based on my partner's setup as the extension worked fine for him), and specifying the updated location in the Python interpreter preference within VS Code. I also checked the box telling the Device Simulator Express Extension to "automatically configure itself for the right dependencies" when the Python interpreter is changed. None of these fixed it. 

The only solution I could find was to use the "Shell Command: Install 'code' command in PATH" within VS Code to install the `code` command line shortcut, and run VS Code from command line. The program, and therefore the simulator, then sees the updated Python environment and the simulator runs successfully.

I have not figured out a way to run VS Code normally and have the simulator work. I can't imagine I'm the only one who has ever run into this, and want to ensure that everyone using this extension has a positive experience. Please let me know if you have any suggestions.

This may be an issue with VS Code and not the simulator. However, if the simulator has a minimum required Python version, this should be well documented. I see in the "Prerequisites" it says Python 3.7+, but it appears to be attempting to use Python 3.7, and the simulator did not work for me. Forcing it to use 3.7.6 was successful.

Tagging @nnja as discussed.

[image]

xnkevinnguyen, 2020-05-06:
This seems to be an issue that happens on MacOS with older python versions and not on the extension side. ( https://github.com/pypa/pipenv/issues/187 https://coderwall.com/p/-k_93g/mac-os-x-valueerror-unknown-locale-utf-8-in-python). Can you confirm that this issue happens on MacOS with version 3.7 only and that it works on 3.7.6?

If that's the case, we might bump up minimum python version. Tagging @andreamah and @vandyliu  in case they have additional input.

kattni, 2020-05-07:
@xnkevinnguyen I am looking deeper into the issue. It turns out it may be an issue on my end, but whatever it is, it's not being handled gracefully by VS Code or the extension. I will let you know what I figure out.

KokoDoko, 2020-07-13:
I get the exact same error message when running the simulator. Also, the device simulator switches to python 3.7.2 automatically. It shows the following message: `Automatically updated interpreter to point to extension's virtual environment`.

nnja, 2020-08-07:
@iennae Are you still the current maintainer? If you are, can you investigate this issue?