microsoft/vscode-python-devicesimulator #373: autocompletion in virtual env

mess-maker, 2020-05-09:
Hi,

not sure if it is the place to ask this kind of question, sorry I'm new to github

I've created a project directory
created a python venv inside the directory
activated the venv
created a new .py file by the command "device simulator express : new file"
(so via vs code inside the venv)
answered yes to the pop up to install the required files

all seems to work, the simulator work the py script runs, but the autocompletion don t work

did I do something wrong ?
maybe i don t need to create a venv by myself ?

thanks for any help and sorry if rise an issue for nothing

andreamah, 2020-05-19:
Hi @mess-maker ! You don't need to create a venv by yourself, since the extension should make one for you if you don't have one. 

If you decide that you want to use your own custom venv, please do the following:
1. [Re-select the Python interpreter](https://code.visualstudio.com/docs/python/environments) to point to your venv's interpreter
2. Restart VS Code or select "install extension dependencies" from the command palette. 

Either way, there is no need to activate the venv 😊.

lukequeenan, 2020-05-24:
I'm seeing a similar problem where `from microbit import *` is unrecognized. However I can find the modules under `from micropython.microbit import *`.

The latter import works for Intellisense and the simulator but I need to change it back to `from microbit import*` for the actual microbit.

erongd, 2020-07-28:
> I'm seeing a similar problem where `from microbit import *` is unrecognized. However I can find the modules under `from micropython.microbit import *`.
> 
> The latter import works for Intellisense and the simulator but I need to change it back to `from microbit import*` when deploying to the actual microbit.

Have exactly the same issue on Mac with latest VSCode and Extensions. The same seems to work on raspberry pi though.

gramster, 2020-08-26:
The extension doesn't use the "real" libraries but has its own "fake" versions that work with the simulator. There's probably a structural difference you are encountering, that we need to fix.