microsoft/vscode-python-devicesimulator #374: Adopt VS Code's 'asWebviewUri' API

mjbvz, 2020-05-16:
Hi, I maintain VS Code's Webview API

## Issue
Our telemetry suggests that your extension uses webviews and may be loading resources in these webviews using hardcoded `vscode-resource:` URIs. These URIs have some important limitations and don't work properly when VS Code is run in a browser. We are also making changes in the desktop version of VS Code that may cause these URIs to not work properly in future versions of VS Code.

While we are making our best effort to continue support existing webview extensions that use `vscode-resource:` URIs on desktop versions of VS Code, we will not able to fully support all uses cases.

## Fix
To ensure that your extension continues to work in as many environments as possible—including on web and remotely—please:

- Switch to use the [`Webview.asWebviewUri` function](https://github.com/microsoft/vscode/blob/307cb32f306a527a30a5b1756e8bf548c141192c/src/vs/vscode.d.ts#L6703) for loading resources.

- Switch to use the [`Webview.cspSource` property](https://github.com/microsoft/vscode/blob/307cb32f306a527a30a5b1756e8bf548c141192c/src/vs/vscode.d.ts#L6714) in content security policies.

These APIs shipped around 2 years ago so they should be available in all modern versions of VS Code. You can find additional information about the issue here: https://github.com/microsoft/vscode/issues/97962

Let me know if you have any questions about this change

xnkevinnguyen, 2020-05-26:
Thank you for reporting the issue. PR #375 addresses this issue.

andreamah, 2020-06-22:
Should be resolved, please open a new issue if there are any further problems.

closed 2020-06-22 by andreamah