microsoft/vscode-python-devicesimulator #377: Installation of Extension Dependencies fails

sskagemo, 2020-07-15:
I have installed the Device Simulator Express Extension, but when trying to install the extension dependencies, it fails with the message:
"Dependency download could not be completed. Functionality may be limited. Please review the installation docs."

I use VS Code on Windows 10, but connected to remote Ubuntu 20.04 on WSL2, Python 3.8.2. I also have the VS Code Python Extension for WSL.

frank4466, 2020-11-11:
Hi i have same problem. I have install licrobit statoed and it's working.

kiransweng, 2020-11-14:
Hi , I have same issue. Windows 10 OS , received error "Dependency download could not be completed. Functionality may be limited. Please review the installation docs. upon running the basic code.
Any workaround?

frank4466, 2020-11-15:
583/5000
Yes and no. I succeeded three times but each time I do so many things that I cannot reproduce it and suddenly it works. 
First I create a working folder, then I install "Device simulator express", then I create a python file in my folder and I install "micro: bit from Statped" which puts libraries in my working folder with the "micro:bit prepare" command. 
I close vscode, I restart it. I uninstall, I reinstall and finally it's works. 
Nothing perfectly reproducible unfortunately. But it is always with these elements that it works.

McSackeri, 2021-01-29:
I was getting the same error with Python 3.9 installed.  Used the VSCode Developer Tools (under Help) to look at the console and saw this:

> RuntimeWarning: Pillow 7.0.0 does not support Python 3.9 and does not provide prebuilt Windows binaries. We do not recommend building from source on Windows.

Edited the file ".vscode\extensions\ms-python.devicesimulatorexpress-2020.0.36321\out\requirements.txt" and changed the line

Pillow==7.0.0
to 
Pillow==8.1.0

Saved the file, restarted VSCode, and ran the dependencies command again, and it completed without errors.

dhust, 2021-02-09:
@McSackeri Where is that file at? In my project's .vscode folder I only see cpx.json and settings.json. Did I not install something that creates these files? Thanks.