microsoft/vscode-python-devicesimulator #415: Unable to install dependencies

FrankSAURET, 2021-10-06:
in the src\requirements.txt file on  github i can see line 8 -> Pillow==8.1.1 but on my PC (i have install device simulator today) there is always -> Pillow==7.0.0

If i manually change Pillow version from 7.0.0 to 8.1.1 i can install the dependecies correctly.

Why ?

jbirbal-skydom, 2021-11-22:
Did you find a solution?

FrankSAURET, 2021-11-22:
Not yet.

jbirbal-skydom, 2021-11-22:
okay fixed it!! 
Change your pillow version in ~\.vscode\extensions\ms-python.devicesimulatorexpress-2020.0.36321\out\requirements.txt to 8.1.0 or whatever

FrankSAURET, 2021-11-23:
I have already done this manually but I am surprised, when I install the plugin, the "requirements.txt" file is not good although the online version is corrected. I am a computer science teacher and I cannot ask my students to do this, they are beginners.